realsenseai/realsense_mipi_platform_driver #167: RealSense MIPI platform driver for MAX96712 DESER on AGX Orin

zvebabi, 2023-07-10:
Hello!
I'm trying to run RealSense D457 camera on AGX Orin.

My hardware environment are:
- Intel RealSense D457 camera * 1
- Nvidia Jetson AGX Orin DevKit (Jetpack5.0.2) * 1
- [Leopard E3653-A03 board](https://www.leopardimaging.com/product/accessories/adapters-carrier-boards/e3653-a03/) (MAX96712) * 1
- Leopard LI-FCB-4T1-SS-2M-NP-T1 cable * 1

In examples D457 is used with GMSL boards based on MAX 9296, but Leopard E3653-A03 board has MAX96712 inside.
Changing device tree is not enough to run D457+MAX96712, because there are some patches to MAX 9296 driver.

Is there any way to get the patch for MAX96712?
Kind regards

dmipx, 2023-07-13:
We had no such system to work with, it is need to be configured in the same way as max9296.

LPXingxing, 2023-07-14:
Hi zvebabi,
We have a reference driver for E3653 and D457, But it is on Xavier not Orin. You can try to modify it to support on Orin.  

 The package is a little large. Can you share me your e-mail address? I can share to you through e-mail.
 
Thanks
Xingxing

zvebabi, 2023-07-14:
Hi @LPXingxing! Here is my email address zvebabi@gmail.com
Thanks a lot

LPXingxing, 2023-07-17:
Hi zvebabi,
Already sent to you. Let me know if you received it.

Thanks
Xingxing

zvebabi, 2023-07-18:
Got it, thank you, Xingxing

benjineeering, 2024-02-15:
Did anyone get any further with the port to Orin?

Nir-Az, 2024-02-18:
@benjineeering WIP.
Check out PR #182

LPXingxing, 2024-02-19:
@zvebabi Have you successfully port the driver Orin?

zvebabi, 2024-02-20:
Unfortunately not, after some experiments with driver we decided to order an off-the-shelf Orin with supporting up to 4pcs RS D457.

benjineeering, 2024-02-20:
@zvebabi which off the shelf Orin did you go for?

dmipx, 2024-02-21:
> Hi zvebabi, We have a reference driver for E3653 and D457, But it is on Xavier not Orin. You can try to modify it to support on Orin.
> 
> The package is a little large. Can you share me your e-mail address? I can share to you through e-mail.
> 
> Thanks Xingxing

I will be happy to recieve sources as well.

LPXingxing, 2024-02-22:
@dmipx Can you share your e-mail address?

dmipx, 2024-02-25:
> @dmipx Can you share your e-mail address?

dmitry.perchanov@intel.com

LPXingxing, 2024-02-27:
> > @dmipx Can you share your e-mail address?
> 
> [dmitry.perchanov@intel.com](mailto:dmitry.perchanov@intel.com)

Let me know if you cannot access the link?

zvebabi, 2024-02-27:
@benjineeering Connectech, they have AGX and NX with custom BSP for RS GMSL. 
I've tested Rudy NX (Part #: ESG605) with two d457 simultaneously

Notion33, 2024-07-04:
Hello @LPXingxing , Can I receive the driver for E3653 and D457? My email address is here - taeyeon.open@gmail.com

arif-ozcan, 2024-07-21:
Hello @LPXingxing,

Could you also share with me?

arif.ozcan@kuartismed.com

Thanks in Advance!

masip85, 2024-07-24:
Hello

Could you share it with me too please? masip85@gmail.com

stfuchs, 2024-08-12:
Hey @LPXingxing any chance you can still send the reference driver?

steffen.fuchs@zebra.com

Thank you.

NobitaYan, 2024-10-17:
Hello

Could you share it with me too please?1358000698@qq.com

cwd502, 2024-10-28:
Could you share it with me too please? cwd100@163.com

sammychentw, 2024-10-30:
Hi @LPXingxing :
Could you share the reference driver for E3653 and D457 with me too please? 
sammychentw@gmail.com

Jp-ctr, 2025-01-01:
Can you please share the reference driver E3653 and d457 with me?
parthijaya73@gmail.com

Jp-ctr, 2025-06-16:
Hello everyone,

We are currently working with the Intel D457 camera in a GMSL setup based on the MAX96712 deserializer, connected to an NVIDIA Jetson AGX Orin running JetPack 5.1.4.

We’re facing an issue where the D457 device intermittently fails to probe after repeatedly rebooting the system over time. When using the i2cdetect command, the camera's slave address appears correctly the first time. However, on subsequent executions, the slave address disappears, and there is always a noticeable delay during the scan.

Has anyone experienced a similar issue with this setup?
Are there any recommended steps or workarounds to ensure the device is reliably detected after reboot?

Any insights or suggestions would be greatly appreciated.

Thanks in advance!

closed 2025-06-16 by ymodlin

Kevin-Liao-Avermedia, 2025-10-16:
Hi @LPXingxing can you please the share the driver with me kevin.liao@avermedia.com

We also purchased the Leopard E3653-A03 board and plan to study MAX96712 + D457 driver porting for the AGX Orin.

Thanks !

tom-robo, 2025-12-09:
Hello, @LPXingxing would it be possible to share the driver with me t0mlane@outlook.com we are also attempting to interface a d457 with Leopard E3653-A03 board. Many thanks.