realsenseai/realsense_mipi_platform_driver #170: Kernel Bug Sometimes When Capturing Metadata

eramirez392, 2023-07-21:
I sometimes see the below Kernel log after restarting depth streaming capture while capturing metadata.
It happens on both JP4.6.1 and JP5.0.2. I used v4l2-ctl to test using these commands.

* Enable metadata catpture
`v4l2-ctl -d /dev/video1 --stream-mmap`

* Enable depth stream capture:
`v4l2-ctl -d /dev/video0  --set-fmt-video=width=1280,height=720,pixelformat=UYVY --stream-mmap`

* Kernel error log
```
[  459.498562] ------------[ cut here ]------------
[  459.498746] WARNING: CPU: 3 PID: 8781 at /home/ANT.AMAZON.COM/raen/work/realsense_mipi_platform_driver-jp4.6/sources_4.6.1/kernel/kernel-4.9/drivers/media/v4l2-core/videobuf2-core.c:905 vb2_buffer_done+0x1f8/0x258
[  459.499093] Modules linked in: bnep fuse xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack br_netfilter zram overlay userspace_alert bluedroid_pm d4xx nvgpu ip_tables x_tables

[  459.499246] CPU: 3 PID: 8781 Comm: vi-output, DS5  Tainted: G        W       4.9.253-tegra #2
[  459.499252] Hardware name: Jetson-AGX (DT)
[  459.499258] task: ffffffc3d091e200 task.stack: ffffffc38ef18000
[  459.499266] PC is at vb2_buffer_done+0x1f8/0x258
[  459.499289] LR is at vi5_capture_dequeue+0x508/0x878
[  459.499298] pc : [<ffffff8008b2dbf0>] lr : [<ffffff8008b452e0>] pstate: 20c00045
[  459.499303] sp : ffffffc38ef1bd00
[  459.499308] x29: ffffffc38ef1bd00 x28: ffffffc3d78a6018 
[  459.499323] x27: 0000000000000000 x26: ffffffc3d7dd4048 
[  459.499337] x25: 0000000000000000 x24: 0000000000000000 
[  459.499352] x23: 0000000000000002 x22: 0000000000000005 
[  459.499366] x21: ffffffc3d78a6930 x20: 00000000000001da 
[  459.499378] x19: ffffffc3ce74bc00 x18: 000000000000153b 
[  459.499392] x17: 0000007f801fc530 x16: 0000000000000001 
[  459.499406] x15: 000000000000317b x14: 0000000000000000 
[  459.499419] x13: 000000000007fa4a x12: 0000000000006c06 
[  459.499433] x11: 0000000000000000 x10: 0000000000000a10 
[  459.499448] x9 : ffffffc38ef1bbd0 x8 : ffffffc3d091ec70 
[  459.499462] x7 : 0000000002e9f800 x6 : 0000000000006c06 
[  459.499475] x5 : 0000000000000400 x4 : 0000000000000000 
[  459.499488] x3 : 0000000000000000 x2 : 0000000000000000 
[  459.499501] x1 : 0000000000000005 x0 : 00000000ffffffff 

[  459.499519] ---[ end trace 49a7388e7c4f58fc ]---
[  459.499619] Call trace:
[  459.499632] [<ffffff8008b2dbf0>] vb2_buffer_done+0x1f8/0x258
[  459.499641] [<ffffff8008b452e0>] vi5_capture_dequeue+0x508/0x878
[  459.499651] [<ffffff8008b45908>] tegra_channel_kthread_capture_dequeue+0x90/0x1c0
[  459.499665] [<ffffff80080db0bc>] kthread+0xec/0xf0
[  459.499678] [<ffffff80080838a0>] ret_from_fork+0x10/0x30
```

Have you seen this issue before?

-Enrique

dmipx, 2023-08-13:
Fixed in the following PR.
https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/jp5-2d457-1-des/kernel/nvidia/5.0.2/0004-Add-vi5_fops-cleanup-code-from-previous-unmerged-PR.patch

You can use dev branch, it's up-to-date with latest features

ymodlin, 2026-11-10:
This was fixed in 2023 (vi5_fops cleanup patch) and has been in every release since. Closing. Please open a new issue if you see it on the current release.

closed 2026-11-10 by ymodlin