realsenseai/realsense_mipi_platform_driver #183: Using two D457 with one deserializer

sancAslan, 2024-02-27:
Hi, I have a question about how the driver handles two D457 cameras and, more specifically, the max9295 serializer.

Looking at the device tree, if I have two D457, it seems like the driver would assign 0x42 and 0x60 to each max9295 respectively.
I presume that the max9295 on each D457 has the same initial i2c address? If that's true, and if both D457s were connected to one max9296, how then does the driver set an independent i2c address for each max9295? Wouldn't there be a bus conflict?

Thanks in advance!

dmipx, 2024-02-28:
Hi.
max9296 has two ports - portA and portB
It has register to switch between ports or set automatic mode.
D457 camera has max9295 serializer with default address 0x40 and asic address 0x10.
When we (device-tree config) initialize camera device, first manually chosen max9296 portA and we reconfigure camera serdes to be different than default address, from 0x40 to 0x60, and reconfigure default asic address from 0x10 to 0x1a
When second agreggated camera device initializing, we manually chose max9296 portB and reconfigure second camera serdes from 0x40 to 0x44 and camera from 0x10 to 0x1b.
This way we have no bus collision.

We have much complicated case with x86 platform, where 8 cameras plugged in with 4 max9296 deserializers on same i2c bus.
With that case, each max9296 should be hardware configured to have different addresses.
On initialization, we setting all max9296 ports to "LINK RESET" state and then one by one we releasing port links and reconfiguring serdes and camera. You can see that procedure [on d4xx.c dev branch](https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/fb890a851c52a608455e20df8e8eb5a744ae71ed/kernel/realsense/d4xx.c#L3263)

masip85, 2024-07-25:
So this comment:
Note: The `--one-cam` and `--dual-cam` option applies only for JetPack 5.0.2,
does it means that if I am into 5.1.2 or 6.0, the driver will be compiled for dual camera by default?

dmipx, 2024-07-28:
> So this comment: Note: The `--one-cam` and `--dual-cam` option applies only for JetPack 5.0.2, does it means that if I am into 5.1.2 or 6.0, the driver will be compiled for dual camera by default?

Hi. We have support for dual camera for 5.1.2 but not for 6.0 yet.
By default, we build one camera configuration. If you want to use dual camera you need to pass that option key.