realsenseai/realsense_mipi_platform_driver #208: The D457 image is abnormal

edu118, 2024-06-06:
We connected D457 to max9296 and then connect to Jetson AGX Orin, but 848*480 resolution images were abnormal, and other resolutions were normal
[image]
[image]

Nir-Az, 2024-06-06:
Hi @edu118 
Cna you share which driver version is used? and which JP?

edu118, 2024-06-06:
D457 Camera firmware version: 5.16.01
JP is 5.1.2

Nir-Az, 2024-06-06:
@edu118 driver version is the last released driver?

edu118, 2024-06-06:
Hi @Nir-Az
yes

Nir-Az, 2024-06-06:
Thanks for the info, we will investigate the issue

edu118, 2024-06-07:
Hi @Nir-Az
Please help to take a look as soon as possible. Our project is a bit urgent, thank you！

Nir-Az, 2024-06-07:
Can you please also share the output of 
modinfo d4xx
And state how many cameras are connected and which deserializer is connected at your setup?

edu118, 2024-06-07:
[image]
4 D457 cameras
2 max9296

edu118, 2024-06-07:
[image]

dmipx, 2024-08-05:
Its resolved with https://github.com/IntelRealSense/realsense_mipi_platform_driver/pull/217

closed 2024-08-05 by dmipx