realsenseai/realsense_mipi_platform_driver #209: Kernel panics by using the ROS CLI tools (rosnode, rostopic) to interact with realsense camera driver

edu118, 2024-06-12:
This produces kernel panics by using the ROS CLI tools (rosnode,
rostopic) to interact with realsense camera driver. The driver has been
running without issues using USB realsenses for multiple years.

Runing this application：
[run_realsense_test.zip](https://github.com/user-attachments/files/15796958/run_realsense_test.zip)
[image]

edu118, 2024-06-12:
The kernel crash log is as follows：
[dmesg.txt](https://github.com/user-attachments/files/15797799/dmesg.txt)

edu118, 2024-06-12:
The streaming mode is "depth-only", so no IR streams.

dmipx, 2024-06-16:
Hi. You can start with driver update.
Together with libRealSense SDK and ROS we upgrade driver functionalities.

edu118, 2024-06-19:
This is the latest drive

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.

If it repeats, also check that the camera isn't in recovery mode (a `/dev/d4xx-dfu*` node without video nodes).