realsenseai/realsense_mipi_platform_driver #215: Step by Step Installation Failed on JetPack 6.0

HappySamuel, 2024-07-11:
**HW: AGX Orin 64GB developer kit
SW: Jetpack 6.0**

I have followed the steps, and shown on below sequence. Do let me know if i do it wrongly.
```
git clone https://github.com/IntelRealSense/realsense_mipi_platform_driver.git
cd realsense_mipi_platform_driver
./setup_workspace.sh 6.0
./apply_patches.sh --dual-cam apply 6.0
sudo apt install build-essential bc flex bison
./build_all.sh 6.0
```
However, it failed when running build_all.sh and it shows error as below:
```
make: Entering directory '/home/agx-orin/Downloads/realsense_mipi_platform_driver/sources_6.0/kernel'
================================================================================
Building kernel-jammy-src sources
================================================================================
make \
	ARCH=arm64 \
	-C /home/agx-orin/Downloads/realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src \
	LOCALVERSION=-tegra \
	defconfig
make[1]: Entering directory '/home/agx-orin/Downloads/realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
*** Default configuration is based on 'defconfig'
/home/agx-orin/Downloads/realsense_mipi_platform_driver/l4t-gcc/6.0/bin/aarch64-buildroot-linux-gnu-gcc: unknown compiler
scripts/Kconfig.include:44: Sorry, this compiler is not supported.
make[2]: *** [scripts/kconfig/Makefile:87: defconfig] Error 1
make[1]: *** [Makefile:630: defconfig] Error 2
make[1]: Leaving directory '/home/agx-orin/Downloads/realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src'
make: *** [Makefile:24: kernel] Error 2
make: Leaving directory '/home/agx-orin/Downloads/realsense_mipi_platform_driver/sources_6.0/kernel
```
Need guidance for how to resolve this step, else couldn't move onto next step. Thank you in advance.

Best,
Samuel

dnovischi, 2024-07-18:
**HW: Jetson Orin Nano 8GB development kit**
**SW: Jetpack 6.0**

Also fails for the above steps for `$ ./build_all_deb.sh 6.0`

```./build_all_deb.sh 6.0
make[1]: Entering directory '../realsense_mipi_platform_driver/images/6.0'
  GEN     Makefile
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
***
*** Can't find default configuration "arch/arm64/configs/tegra_defconfig"!
***
make[2]: *** [.../realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src/scripts/kconfig/Makefile:94: tegra_defconfig] Error 1
make[1]: *** [.../realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src/Makefile:630: tegra_defconfig] Error 2
make[1]: Leaving directory '../realsense_mipi_platform_driver/images/6.0'
make: *** [Makefile:226: __sub-make] Error 2

```

Nir-Az, 2024-07-18:
We have an updated guide on the dev branch.
Did you try following this steps?
https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/dev/README_JP6.md

Download Toolchain ARM64 compiler - [Bootlin Toolchain gcc 11.3](https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v3.0/toolchain/aarch64--glibc--stable-2022.08-1.tar.bz2)

Here you can see how the flow works on out GitHub Actions flow - https://github.com/IntelRealSense/realsense_mipi_platform_driver/actions/runs/9955168664/workflow

 > Not sure dual camera support was tested yet on JP6

dnovischi, 2024-07-19:
Thank you.
I will try it out and come back after the weekend

dmipx, 2024-07-28:
> **HW: Jetson Orin Nano 8GB development kit** **SW: Jetpack 6.0**
> 
> Also fails for the above steps for `$ ./build_all_deb.sh 6.0`
> 
Hi,
First of all - we never verified Orin Nano.

We not suport yet debian build for jp6.

dmipx, 2024-07-28:
> Thank you. I will try it out and come back after the weekend
> 
> Update: @Nir-Az I have tried it twice with no luck. The development branch tutorial fails on the _JetPack manual build - cross compile x86-64 (CI deploy)_ step with the following:

Please find error statement in output.

dnovischi, 2024-07-28:
@dmipx Thank you for your comments.

This means that none of D4xx production drivers support any of the Jetson platforms for the 5.15 kernel (Ubuntu 22.04 roofs for JP6).

Thus hobbiest may use the RSUSB with some errors for ROS2 humble, Isaac Sim and so on. But at this time **none of D4xx is recommended to be used in any product.**

Nir-Az, 2024-07-28:
Hi @dnovischi ,
JP6 support is still WIP.
We started with a single camera support, multi camera support will be added ASAP as well.
You can keep track of [dev](https://github.com/IntelRealSense/realsense_mipi_platform_driver/tree/dev) branch to get latest updates once they arrive.

dnovischi, 2024-07-28:
@Nir-Az Thanks. Will do, hope this gets resolved soon.

masip85, 2024-07-29:
I am having this issue too. Fails with cross compile, but also natively from arm64, orin nano or agx orin..  Dev or main branch. Both of them fails with 6.0. Just like:
https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/215#issuecomment-2238732226

What can I do? 

By the way,if I try to use the automatic way,this happens:

```
Press any key to continue (or wait 30 seconds..)
--2024-07-29 18:25:32--  https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v3.0/toolchain/aarch64--glibc--stable-2022.08-1.tar.bz2
Resolviendo developer.nvidia.com (developer.nvidia.com)... 152.199.20.126
Conectando con developer.nvidia.com (developer.nvidia.com)[152.199.20.126]:443... conectado.
Petición HTTP enviada, esperando respuesta... 302 Found
Ubicación: https://developer.download.nvidia.com/embedded/L4T/r36_Release_v3.0/toolchain/aarch64--glibc--stable-2022.08-1.tar.bz2 [siguiente]
--2024-07-29 18:25:33--  https://developer.download.nvidia.com/embedded/L4T/r36_Release_v3.0/toolchain/aarch64--glibc--stable-2022.08-1.tar.bz2
Resolviendo developer.download.nvidia.com (developer.download.nvidia.com)... 152.199.20.126
Conectando con developer.download.nvidia.com (developer.download.nvidia.com)[152.199.20.126]:443... conectado.
Petición HTTP enviada, esperando respuesta... 200 OK
Longitud: 140109376 (134M) [application/x-bzip]
Guardando como: ‘aarch64--glibc--stable-final.tar.bz2’

aarch64--glibc--stable-final. 100%[================================================>] 133,62M  64,9MB/s    en 2,1s    

2024-07-29 18:25:35 (64,9 MB/s) - ‘aarch64--glibc--stable-final.tar.bz2’ guardado [140109376/140109376]

Downloading default kernel/kernel-jammy-src source...
Clonando en 'sources_6.0/kernel/kernel-jammy-src'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 34.216.216.36]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 35.82.181.43]: errno=Expiró el tiempo de conexión

sources_6.0/kernel/kernel-jammy-src source sync failed!

Downloading default nvgpu source...
Clonando en 'sources_6.0/nvgpu'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 34.216.216.36]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 35.82.181.43]: errno=Expiró el tiempo de conexión

sources_6.0/nvgpu source sync failed!

Downloading default nvidia-oot source...
Clonando en 'sources_6.0/nvidia-oot'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 35.82.181.43]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 34.216.216.36]: errno=Expiró el tiempo de conexión

sources_6.0/nvidia-oot source sync failed!

Downloading default hwpm source...
Clonando en 'sources_6.0/hwpm'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 35.82.181.43]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 34.216.216.36]: errno=Expiró el tiempo de conexión

sources_6.0/hwpm source sync failed!

Downloading default nvethernetrm source...
Clonando en 'sources_6.0/nvethernetrm'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 34.216.216.36]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 35.82.181.43]: errno=Expiró el tiempo de conexión

sources_6.0/nvethernetrm source sync failed!

Downloading default hardware/nvidia/t23x/nv-public source...
Clonando en 'sources_6.0/hardware/nvidia/t23x/nv-public'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 35.82.181.43]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 34.216.216.36]: errno=Expiró el tiempo de conexión

sources_6.0/hardware/nvidia/t23x/nv-public source sync failed!

Downloading default hardware/nvidia/tegra/nv-public source...
Clonando en 'sources_6.0/hardware/nvidia/tegra/nv-public'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 34.216.216.36]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 35.82.181.43]: errno=Expiró el tiempo de conexión

sources_6.0/hardware/nvidia/tegra/nv-public source sync failed!

Downloading default nvdisplay source...
Clonando en 'sources_6.0/nvdisplay'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 34.216.216.36]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 35.82.181.43]: errno=Expiró el tiempo de conexión

sources_6.0/nvdisplay source sync failed!

Downloading default dtc-src/1.4.5 source...
Clonando en 'sources_6.0/dtc-src/1.4.5'...
fatal: no es posible conectar a nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 34.216.216.36]: errno=Expiró el tiempo de conexión
nv-tegra.nvidia.com[1: 35.82.181.43]: errno=Expiró el tiempo de conexión

sources_6.0/dtc-src/1.4.5 source sync failed!

Downloading default tegra/argus-cam-libav/argus_cam_libavencoder source...
Clonando en 'sources_6.0/tegra/argus-cam-libav/argus_cam_libavencoder'...
^C
```

Nir-Az, 2024-07-30:
Hi @masip85 ,
Your log show internet access issues, could be you are behind a proxy server?

Try running this on your web browser and see if you can download the file (works for me)
`https://developer.download.nvidia.com/embedded/L4T/r36_Release_v3.0/toolchain/aarch64--glibc--stable-2022.08-1.tar.bz2`

dmipx, 2024-08-05:
> sources_6.0/kernel/kernel-jammy-src source sync failed!
It happens when you're under proxy.
Try to verify proxy settings.
Another workaround is to substitute git: links with https: links.
`git config --global url."https://nv-tegra.nvidia.com/".insteadOf "git://nv-tegra.nvidia.com/"`
`git config --global url."https://github.com/".insteadOf "git@github.com/"`

SprAJR, 2024-08-13:
> ```
> git clone https://github.com/IntelRealSense/realsense_mipi_platform_driver.git
> cd realsense_mipi_platform_driver
> ./setup_workspace.sh 6.0
> ./apply_patches.sh --dual-cam apply 6.0
> sudo apt install build-essential bc flex bison
> ./build_all.sh 6.0
> ```

I was having the same issue.  doing 'git checkout dev' after git clone on the manual build worked.  Now I'm trying to figure out why it killed my USB keyboard/mouse driver... but hey at least the cameras work!  (two D455 on Nvidia Orin NX each with D4 board, connected by two USB plugs, Jetpack 6.0)

AG-TM, 2024-08-24:
same issue here, it didn't work with jp6 followings the steps

Nir-Az, 2024-08-29:
We had a commit on dev branch with debug symbols default on, this require the user to replace all kernel modules and might explan the keyboard mouse issue.
We now changed it back to default off.

Another thing is that the manual on dev has steps that clone the master (this is on purpose as this will be pushed to master on the next release)

Please retry on latest dev manual and code and replace all checkout steps with `-b Dev` parameter

Hope it helps

masip85, 2024-09-10:
I am giving it a try with orin Nano. I am hopeful, because of that Orin NX recognising it.  Questions:

Should I modify this line:
   cp $SRCS/nvidia-oot/device-tree/platform/generic-dts/dtbs/tegra234-p3737-0000+p3701-0000-nv.dtb $TEGRA_KERNEL_OUT/rootfs/boot/dtb/
   
@Nir-Az, I guess that until  this point, every dtb has been patched, and here you are selecting the AGX one, isn't it?
Is there more points to modify for Orin Nano? I'd like to make a try. [see here](https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/221)

@SprAJR , did you modified this? If no, how is possible this is working for your NX's without modifying this line?

2805651606, 2024-10-13:
请问，你们有人解决了？我的板子是orin nx

Nir-Az, 2024-10-13:
@Arun-Prasad-V  can you take a look on @masip85  comment here:

https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/215#issuecomment-2340828668

closed 2025-06-25 by ymodlin