realsenseai/realsense_mipi_platform_driver #221: Orin Nano support

masip85, 2024-07-31:
Hello

I've read Orin Nano is not supported. I've tried to use this project with 5.3 and 6.0 versions without success. Why is that? Can I  modify something to  achieve it?

Nir-Az, 2024-07-31:
Jetson Nano is not "Officially" supported because it is not tested on our side.
As I see, many users from the community use it.
I recommend going over https://github.com/IntelRealSense/librealsense/issues

masip85, 2024-07-31:
But what I am seeing at the issues of librealsense is  people who is using GSML with embedded (normally Connect Tech solutions).  But I am using 9296 deserializer board.  

I've found that at step:
```
With Jetson-IO tool: sudo /opt/nvidia/jetson-io/jetson-io.py

    Configure Jetson AGX CSI Connector
    Configure for compatible hardware
    Jetson RealSense Camera D457

With command line

sudo /opt/nvidia/jetson-io/config-by-hardware.py -n 2="Jetson RealSense Camera D457"
```
the output is:
NameError: No configuration found for Jetson RealSense Camera D457 on Jetson 24pin CSI Connector!

What should I modify to test it?

masip85, 2024-09-26:
Ok. I've tried modifying this:

build_all.sh
```
-    cp $SRCS/nvidia-oot/device-tree/platform/generic-dts/dtbs/tegra234-p3737-0000+p3701-0000-nv.dtb $TEGRA_KERNEL_OUT/rootfs/boot/dtb/
+    cp $SRCS/nvidia-oot/device-tree/platform/generic-dts/dtbs/tegra234-p3768-0000+p3767-0005-nv.dtb $TEGRA_KERNEL_OUT/rootfs/boot/dtb/
```

Also I've avoided the headers with the pin configuration of the Orin, and I've changed them for the headers I believe are usable for :

```
diff --git a/hardware/realsense/tegra234-camera-d4xx-overlay.dts b/hardware/realsense/tegra234-camera-d4xx-overlay.dts
index 42b2cef..aee4eff 100644
--- a/hardware/realsense/tegra234-camera-d4xx-overlay.dts
+++ b/hardware/realsense/tegra234-camera-d4xx-overlay.dts
@@ -6,7 +6,7 @@
 
 #include <dt-bindings/clock/tegra234-clock.h>
 #include <dt-bindings/gpio/tegra234-gpio.h>
-#include <dt-bindings/tegra234-p3737-0000+p3701-0000.h>
+#include <dt-bindings/tegra234-p3767-0000-common.h>

```
```
diff --git a/hardware/realsense/tegra234-camera-d4xx-overlay-dual.dts b/hardware/realsense/tegra234-camera-d4xx-overlay-dual.dts
index 88f82ee..10f06df 100644
--- a/hardware/realsense/tegra234-camera-d4xx-overlay-dual.dts
+++ b/hardware/realsense/tegra234-camera-d4xx-overlay-dual.dts
@@ -6,7 +6,8 @@
 
 #include <dt-bindings/clock/tegra234-clock.h>
 #include <dt-bindings/gpio/tegra234-gpio.h>
-#include <dt-bindings/tegra234-p3737-0000+p3701-0000.h>
+#include <dt-bindings/tegra234-p3767-0000-common.h>

```

I've been able to compile everything, install (copy). Even shows up in jetson-io.py and let me change it to the new configuration. BUT, after reboot, which boot properly by the way, nothing else happens. dmseg output shows nothing when I grep by d4xx. grep by tegra-capture-vi only showing three lines.

I've double checked dtb and Image has been properly copied.

I am using a Leopard Max9296 deserializer.

Arun-Prasad-V, 2024-10-15:
@masip85, could you please check if I2C mux is detected or not? https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/dev/README_JP6.md#known-issues

masip85, 2024-10-16:
The Orin Nano I mentioned in previous post 
`I've been able to compile everything, install (copy). Even shows up in jetson-io.py and let me change it to the new configuration.`

At this state:
```
vmasip@nano:~$ sudo dmesg | grep i2c

[    0.253464] i2c_dev: i2c /dev entries driver
[    4.102931] tegra-i2c 3160000.i2c: Adding to iommu group 1
[    4.106675] tegra-i2c 3180000.i2c: Adding to iommu group 1
[    4.109401] tegra-i2c 31b0000.i2c: Adding to iommu group 1
[    4.114809] tegra-i2c c240000.i2c: Adding to iommu group 1
[    4.119839] i2c 1-0025: Fixing up cyclic dependency with 3520000.padctl
[    4.120643] tegra-i2c c250000.i2c: Adding to iommu group 1
[    4.126627] i2c i2c-2: Added multiplexed i2c bus 9
[    4.128382] i2c i2c-2: Added multiplexed i2c bus 10
[    4.128402] i2c-mux-gpio bus@0:cam_i2cmux: 2 port mux on 3180000.i2c adapter
[   14.229907] imx219 9-0010: imx219_board_setup: error during i2c read probe (-121)
[   14.258674] imx219 10-0010: imx219_board_setup: error during i2c read probe (-121)
vmasip@nano:~$ sudo dmesg | grep pca954x
vmasip@nano:~$ 
```

Just in case, I paste you other Orin Nano (I've got it with JP5) normal message when you grep i2c is:
```
[    4.592960] i2c /dev entries driver
[    4.596699] tegra-i2c 3160000.i2c: Adding to iommu group 2
[    4.602505] tegra-i2c c240000.i2c: Adding to iommu group 2
[    4.608205] tegra-i2c 3180000.i2c: Adding to iommu group 2
[    4.613908] tegra-i2c 3190000.i2c: Adding to iommu group 2
[    4.619594] tegra-i2c 31b0000.i2c: Adding to iommu group 2
[    4.625277] tegra-i2c 31c0000.i2c: Adding to iommu group 2
[    4.630960] tegra-i2c c250000.i2c: Adding to iommu group 2
[    4.636650] tegra-i2c 31e0000.i2c: Adding to iommu group 2
[    8.781618] i2c i2c-2: Added multiplexed i2c bus 9
[    8.787310] i2c i2c-2: Added multiplexed i2c bus 10
[    8.792334] i2c-mux-gpio cam_i2cmux: 2 port mux on 3180000.i2c adapter
[   15.404324] imx219 9-0010: imx219_board_setup: error during i2c read probe (-121)
[   15.471386] imx219 10-0010: imx219_board_setup: error during i2c read probe (-121)
```

Arun-Prasad-V, 2024-10-16:
@masip85, looks like some i2c misconfiguration in device tree only. Could you please go over the forums and see if any of the conversations helps debugging the issue?
https://forums.developer.nvidia.com/t/mipi-csi2-cam-i2cmux-i2c-2-duplicates-my-i2c-0-entry/182583
https://forums.developer.nvidia.com/t/imx219-board-setup-error-during-i2c-read-probe-121/160230

masip85, 2024-10-16:
First of all:
 I guess error during i2c read probe is not an issue. Imx219 sensor is not related to realsense driver. Moreover, this error appears too with default JP&kernel. Am I right? So what's the point of second link:

Just in case, I've made a check and: 
many relate "error during i2c read probe" to hardware connection problems too (not only, obviously). So, I've moved from one physical  csi port to another.  Changing ribbon csi cable too. The only thing I see changing is that now I see something connected  from bus 10 to bus 9:
```
vmasip@nano:~$ sudo i2cdetect -r -y 9
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```
By the way, no "UU" here in both buses. All logs are shown equal as before.

Regards the first link,  I don't know what's the point you are looking for. Do you talk about this:
https://forums.developer.nvidia.com/t/mipi-csi2-cam-i2cmux-i2c-2-duplicates-my-i2c-0-entry/182583/4?u=masip85
and changing [this](https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/315b6aecce27cccb5b5441879b3baa208f0ac9d1/hardware/realsense/tegra194-camera-d4xx.dtsi#L41) line

Arun-Prasad-V, 2024-10-17:
@masip85, I understand these forums are not directly related to this issue. They talk about few debugging steps. I meant to look at those and check if they can help narrowing down the issue.

```
I guess error during i2c read probe is not an issue. Imx219 sensor is not related to realsense driver. 
Moreover, this error appears too with default JP&kernel. Am I right?
```
-- Not sure. We have never tried with Orin Nano. 

```
Regards the first link, I don't know what's the point you are looking for. Do you talk about this:
https://forums.developer.nvidia.com/t/mipi-csi2-cam-i2cmux-i2c-2-duplicates-my-i2c-0-entry/182583/4?u=masip85
and changing this line?
```
-- Yes, if that's applicable & possible based on your device tree.

What do you observe from below commands?
_sudo i2cdetect -l
v4l2-ctl --list-devices_

masip85, 2024-10-17:
-  sudo i2cdetect -l

[sudo] password for vmasip: 
i2c-0	i2c       	3160000.i2c                     	I2C adapter
i2c-1	i2c       	c240000.i2c                     	I2C adapter
i2c-2	i2c       	3180000.i2c                     	I2C adapter
i2c-4	i2c       	Tegra BPMP I2C adapter          	I2C adapter
i2c-5	i2c       	31b0000.i2c                     	I2C adapter
i2c-7	i2c       	c250000.i2c                     	I2C adapter
i2c-9	i2c       	i2c-2-mux (chan_id 1)           	I2C adapter
i2c-10	i2c       	i2c-2-mux (chan_id 0)           	I2C adapter
i2c-11	i2c       	NVIDIA SOC i2c adapter 0        	I2C adapter

-   v4l2-ctl --list-devices

Cannot open device /dev/video0, exiting.

- By the way, maybe the driver is not loaded? You've got more lines when you execute this. 
```
vmasip@nano11:~$ sudo dmesg | grep tegra-capture-vi
[   11.638731] platform 13e00000.host1x:nvcsi@15a00000: Fixing up cyclic dependency with tegra-capture-vi
[   12.749897] tegra-camrtc-capture-vi tegra-capture-vi: subdev 13e00000.host1x:nvcsi@15a00000- bound
[   12.749915] tegra-camrtc-capture-vi tegra-capture-vi: subdev 13e00000.host1x:nvcsi@15a00000- bound
```

I am checking the device tree and it's dependencies, and I really don't know if that line must be changed. I've got a lack of understanding of device trees.

masip85, 2024-10-28:
I've been trying to obtain [help](https://forums.developer.nvidia.com/t/error-i2c-read-probe/310155/6) from nvidia, but didn't work out. Do you have any clue I can try?

chenx915, 2024-12-05:
@masip85 ，I have the same problem as you. Have you solved it?

chenx915, 2024-12-06:
@masip85 I hope it will be helpful to you. I am currently also in the ORIN NX series, using the max9296 de-serializer. Now my device can see the d4xx dmesg information.
`nvidia@nvidia-desktop:~$ sudo dmesg | grep d4xx
[sudo] password for nvidia:
[    6.399305] d4xx 9-001a: Probing driver for D45x
[    6.399418] d4xx 9-001a: board setup failed
[    6.399517] d4xx 10-001a: Probing driver for D45x
[    6.399591] d4xx 10-001a: board setup failed
[    6.399647] d4xx 14-001a: Probing driver for D45x
[    6.399705] d4xx 14-001a: board setup failed
[    6.399771] d4xx 15-001a: Probing driver for D45x
[    6.399835] d4xx 15-001a: board setup failed
[    7.963039] d4xx 9-001a: Probing driver for D45x
[    7.964435] d4xx 9-001a: board setup failed
[    7.969534] d4xx 10-001a: Probing driver for D45x
[    7.974557] d4xx 10-001a: board setup failed
[    7.989020] d4xx 14-001a: Probing driver for D45x
[    7.994666] d4xx 14-001a: board setup failed
[    8.012000] d4xx 15-001a: Probing driver for D45x
[    8.014406] d4xx 15-001a: board setup failed
[    8.017958] d4xx 9-001a: Probing driver for D45x
[    8.023362] d4xx 9-001a: board setup failed
[    8.026403] d4xx 10-001a: Probing driver for D45x
[    8.027963] d4xx 10-001a: board setup failed
[    8.033566] d4xx 14-001a: Probing driver for D45x
[    8.035486] d4xx 14-001a: board setup failed
[    8.047716] d4xx 15-001a: Probing driver for D45x
[    8.047829] d4xx 15-001a: board setup failed
[    8.050802] d4xx 9-001a: Probing driver for D45x
[    8.054550] d4xx 9-001a: board setup failed
[    8.055191] d4xx 10-001a: Probing driver for D45x
[    8.057056] d4xx 10-001a: board setup failed
[    8.060456] d4xx 14-001a: Probing driver for D45x
[    8.062420] d4xx 14-001a: board setup failed
[    8.062850] d4xx 15-001a: Probing driver for D45x
[    8.062959] d4xx 15-001a: board setup failed
[    8.137897] d4xx 9-001a: Probing driver for D45x
[    8.143632] d4xx 9-001a: board setup failed
[    8.146921] d4xx 10-001a: Probing driver for D45x
[    8.147162] d4xx 10-001a: board setup failed
[    8.147524] d4xx 14-001a: Probing driver for D45x
[    8.147637] d4xx 14-001a: board setup failed
[    8.147757] d4xx 15-001a: Probing driver for D45x
[    8.147819] d4xx 15-001a: board setup failed
[    8.242504] d4xx 9-001a: Probing driver for D45x
[    8.242623] d4xx 9-001a: board setup failed
[    8.242820] d4xx 10-001a: Probing driver for D45x
[    8.242917] d4xx 10-001a: board setup failed
[    8.243072] d4xx 14-001a: Probing driver for D45x
[    8.243136] d4xx 14-001a: board setup failed
[    8.243276] d4xx 15-001a: Probing driver for D45x
[    8.243328] d4xx 15-001a: board setup failed
[   10.119676] d4xx 9-001a: Probing driver for D45x
[   10.119808] d4xx 9-001a: board setup failed
[   10.120062] d4xx 10-001a: Probing driver for D45x
[   10.120139] d4xx 10-001a: board setup failed
[   10.120449] d4xx 14-001a: Probing driver for D45x
[   10.120555] d4xx 14-001a: board setup failed
[   10.123353] d4xx 15-001a: Probing driver for D45x
[   10.123839] d4xx 15-001a: board setup failed
nvidia@nvidia-desktop:~$ sudo dmesg | grep i2c
[    0.179492] i2c_dev: i2c /dev entries driver
[    1.743726] tegra-i2c 3160000.i2c: Adding to iommu group 1
[    1.749405] tegra-i2c 3180000.i2c: Adding to iommu group 1
[    1.752271] tegra-i2c 31b0000.i2c: Adding to iommu group 1
[    1.755433] tegra-i2c c240000.i2c: Adding to iommu group 1
[    1.757264] i2c 1-0025: Fixing up cyclic dependency with 3520000.padctl
[    1.757852] tegra-i2c c250000.i2c: Adding to iommu group 1
[    1.759666] i2c i2c-2: Added multiplexed i2c bus 9
[    1.759903] i2c i2c-2: Added multiplexed i2c bus 10
[    1.760004] i2c i2c-2: Added multiplexed i2c bus 11
[... 176 lines cut ...]
-                                                               vcc-supply = <&vdd_1v8_ls>;
+                                                               vcc-supply = <&vdd_1v8_sys>;
                                                                cam-type = "Y8";
                                                                nvidia,gmsl-ser-device = <&ser_a>;
                                                                nvidia,gmsl-dser-device = <&dser>;
@@ -428,7 +439,7 @@
                                                                reg = <0x1a>;
                                                                compatible = "intel,d4xx";
                                                                use_sensor_mode_id = "true";
-                                                               vcc-supply = <&vdd_1v8_ls>;
+                                                               vcc-supply = <&vdd_1v8_sys>;
                                                                cam-type = "IMU";
                                                                nvidia,gmsl-ser-device = <&ser_a>;
                                                                nvidia,gmsl-dser-device = <&dser>;
@@ -469,7 +480,7 @@
                                                        };
                                                };
                                        };
-                               };
+                               //};
                        };
                };
        };
`
Before the modification, I had the same problem with you.

chenx915, 2024-12-06:
[image]
[image]
[image]
[image]
Before the modification, I had the same problem with you.
After the modification, I can query the d4xx related information in dmesg. Perhaps we can go a step further and adapt to the prompts.
[image]

Arun-Prasad-V, 2024-12-18:
Hi @masip85, since you are using Leopard deserializer, please take a look at this https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/166 similar conversation on AGX Xavier and see if it can help.

masip85, 2025-02-10:
Nowadays, I don't have the leopard anymore. But we are testing with other cameras:
https://www.waveshare.com/wiki/MAX9296-GMSL-Camera-Board#Driver

It should work too, because it's MAX9296, but I have same error as before. 

@chenx915, I have tried this with same results. I've got no good step forwards for you sorry:
> [image] [image] [image] [image] Before the modification, I had the same problem with you. After the modification, I can query the d4xx related information in dmesg. Perhaps we can go a step further and adapt to the prompts. [image]

But I've seen you've already gone forward but you've got another problem:
https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/246

@Arun-Prasad-V , I think they are waiting for your help there at that issue.

Arun-Prasad-V, 2025-02-10:
> Hi [@masip85](https://github.com/masip85), since you are using Leopard deserializer, please take a look at this [#166](https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/166) similar conversation on AGX Xavier and see if it can help.

@masip85, @chenx915, as suggested before (https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/166#issuecomment-1633732254), if you are **not** using Intel's GMSL/FAKRA De-Serializer board, could you try shorting I2C lines?

chenx915, 2025-02-11:
> > Hi [@masip85](https://github.com/masip85), since you are using Leopard deserializer, please take a look at this [#166](https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/166) similar conversation on AGX Xavier and see if it can help.
> 
> [@masip85](https://github.com/masip85), [@chenx915](https://github.com/chenx915), as suggested before ([#166 (comment)](https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/166#issuecomment-1633732254)), if you are **not** using Intel's GMSL/FAKRA De-Serializer board, could you try shorting I2C lines?

As required by the company, other solutions have been replaced in the project, and D457 has not been adapted for the time being.

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.