realsenseai/realsense_mipi_platform_driver #235: make: *** [Makefile:1912: drivers] Error 2

nikk-source, 2024-09-23:
make: *** [Makefile:1912: drivers] Error 2
make: *** Waiting for unfinished jobs....
I got this, after make -j$(nproc) compilation, i got this error. So, How can i resolve it.

demisrael, 2024-09-23:
Try with `./build_all.sh`

dmipx, 2024-09-23:
Try with `./build_all.sh`

nikk-source, 2024-09-24:
Not working.

Nir-Az, 2024-09-24:
Hi @nikk-source ,
Please attach the full log, if you are under a proxy make sure the script is able to access the servers it needs.

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.

The build scripts have been reworked since. Please retry with `setup_workspace.sh` → `apply_patches.sh` → `build_all.sh` and attach the full log if it fails.