realsenseai/realsense_mipi_platform_driver #238: ORIN NX Jetpack 6.0 install

RoboMike72, 2024-10-23:
Is the only way to setup the MIPI driver by installing and compiling on a host computer then copying it to the ORIN? Is there a method that can be done only on the ORIN? I am trying to make an installation script for 4 of them.

Also, after my last install I lost keyboard and mouse control too.  Is there a workaround to prevent this?  I noticed it in another post too.

Arun-Prasad-V, 2024-10-28:
> Is the only way to setup the MIPI driver by installing and compiling on a host computer then copying it to the ORIN? Is there a method that can be done only on the ORIN? I am trying to make an installation script for 4 of them.

We would suggest you to use a host machine for building as it is developed and tested there. Same steps might work on Orin as well. You could try and check.

> Also, after my last install I lost keyboard and mouse control too. Is there a workaround to prevent this? I noticed it in another post too.

Did you copy the kernel `Image`? If so, you have to copy and load the corresponding kernel modules as well to be compatible (Meaning, you have to copy the "kernel" folder as well and follow the similar steps mentioned for loading the modules in "extra" and "updates" folders). 

1. Copy `images/6.0/rootfs/lib/modules/5.15.136-tegra/kernel` from host to `/lib/modules/5.15.136-tegra/` on Orin target
2. To enable depmod to scan `kernel` folder, add an entry 'kernel' in the file `/etc/depmod.d/ubuntu.conf`
3. Run `sudo depmod`
 
Note: For JP6, if the kernel is unmodified, you don't have to copy the `Image` and load `kernel` modules. Loading the modules in `extra` and `updates` folders are enough.

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.

Building on a host machine is still the supported path.

RoboMike72, 2026-11-10:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Unfortunately, that project died off.  I have no urgency for this right
now. If you still want it tested let me know and I will try to find the
time.  Otherwise go ahead and close it.