realsenseai/realsense_mipi_platform_driver #239: Issue with Metadata Capture on Jetson Xavier AGX Devkit JetPack 5.1.2

rr-jmorera, 2024-10-24:
Using a Jetson Xavier AGX devkit with JetPack version 5.1.2 I'm encountering an issue when I enable metadata capture using the following command:

`v4l2-ctl -d /dev/video2 --stream-mmap`

This problem occurs with multiple devices, specifically /dev/video2 (RGB Stream), /dev/video4 (IR Stream), and /dev/video5 (IMU Stream). The capture itself seems to work without any issues, but the kernel log contains the following error messages related to the metadata capture:

```
[ 1650.513874] tegra-camrtc-capture-vi tegra-capture-vi: corr_err: discarding frame 0, flags: 0, err_data 131072, vc: 1
[ 1650.515655] ------------[ cut here ]------------
[ 1650.515708] WARNING: CPU: 3 PID: 3829 at /home/jmorera/work/devdirs/SymboticRealsense/realsense_mipi_platform_driver/Linux_for_Tegra/source/public/kernel/nvidia/drivers/media/platform/tegra/camera/fusa-capture/capture-vi-channel.c:274 vi_channel_close_ex+0x7c/0x90
[ 1650.516280] Modules linked in: fuse(E) xt_conntrack(E) xt_MASQUERADE(E) nf_conntrack_netlink(E) nfnetlink(E) iptable_nat(E) nf_nat(E) nf_conntrack(E) nf_defrag_ipv6(E) nf_defrag_ipv4(E) libcrc32c(E) xt_addrtype(E) iptable_filter(E) br_netfilter(E) lzo_rle(E) lzo_compress(E) zram(E) overlay(E) ramoops(E) reed_solomon(E) snd_soc_tegra210_ope(E) snd_soc_tegra186_dspk(E) snd_soc_tegra186_asrc(E) snd_soc_tegra210_mvc(E) snd_soc_tegra186_arad(E) snd_soc_tegra210_iqc(E) snd_soc_tegra210_afc(E) snd_soc_tegra210_dmic(E) snd_soc_tegra210_adx(E) snd_soc_tegra210_amx(E) snd_soc_tegra210_mixer(E) snd_soc_tegra210_i2s(E) snd_soc_tegra210_admaif(E) snd_soc_tegra210_sfc(E) snd_soc_tegra_pcm(E) aes_ce_blk(E) crypto_simd(E) cryptd(E) aes_ce_cipher(E) ghash_ce(E) sha2_ce(E) sha256_arm64(E) sha1_ce(E) snd_soc_spdif_tx(E) snd_soc_tegra_machine_driver(E) ucsi_ccg(E) typec_ucsi(E) typec(E) snd_hda_codec_hdmi(E) snd_soc_tegra210_adsp(E) snd_hda_tegra(E) snd_soc_rt5659(E) snd_soc_tegra_utils(E) snd_hda_codec(E)
[ 1650.516660]  nct1008(E) snd_soc_simple_card_utils(E) snd_soc_rl6231(E) nvadsp(E) userspace_alert(E) bluedroid_pm(E) tegra210_adma(E) max77620_thermal(E) snd_soc_tegra210_ahub(E) snd_hda_core(E) tegra_bpmp_thermal(E) ahci(E) libahci(E) loop(E) libata(E) spi_tegra114(E) ina3221(E) pwm_fan(E) nvgpu(E) nvmap(E) binfmt_misc(E) d4xx(E) ip_tables(E) x_tables(E) [last unloaded: mtd]
[ 1650.516821] CPU: 3 PID: 3829 Comm: vi-output, DS5  Tainted: G        W  OE     5.10.120-tegra #1
[ 1650.516828] Hardware name: Unknown Jetson-AGX/Jetson-AGX, BIOS 4.1-33958178 08/01/2023
[ 1650.516844] pstate: 80c00009 (Nzcv daif +PAN +UAO -TCO BTYPE=--)
[ 1650.516857] pc : vi_channel_close_ex+0x7c/0x90
[ 1650.516866] lr : vi_channel_close_ex+0x3c/0x90
[ 1650.516872] sp : ffff80001afcbcf0
[ 1650.516880] x29: ffff80001afcbcf0 x28: ffff02de16c0a080 
[ 1650.516896] x27: 0000000000000000 x26: ffff02de16c0a880 
[ 1650.516912] x25: ffff02db05930000 x24: 0000000000000000 
[ 1650.516927] x23: ffff02dad47bc3c4 x22: ffff02dac372da80 
[ 1650.516943] x21: ffff02dac372daa0 x20: 0000000000000001 
[ 1650.516958] x19: ffff02dacaf25500 x18: 0000000000000000 
[ 1650.516973] x17: 0000000000000000 x16: 0000000000000000 
[ 1650.517022] x15: 0000000000000000 x14: 0000000000000000 
[ 1650.517038] x13: 0000000000000001 x12: 00000000000005d2 
[ 1650.517055] x11: 0000000000000040 x10: 0000000000000ab0 
[ 1650.517073] x9 : 00000000fffffffe x8 : ffff02db05930000 
[ 1650.517090] x7 : ffff000000000000 x6 : 0000000000000001 
[ 1650.517152] x5 : 0000000000000001 x4 : fffffe0b6b31ee20 
[ 1650.517167] x3 : 0000000000000000 x2 : ffff02db05930000 
[ 1650.517183] x1 : ffff02dac372da88 x0 : ffff02daeae43400 
[ 1650.517201] Call trace:
[ 1650.517230]  vi_channel_close_ex+0x7c/0x90
[ 1650.517282]  vi5_capture_dequeue+0x5e8/0x740
[ 1650.517291]  tegra_channel_kthread_capture_dequeue+0xb0/0x1f0
[ 1650.517308]  kthread+0x148/0x170
[ 1650.517323]  ret_from_fork+0x10/0x24
[ 1650.517332] ---[ end trace 2d9afa0fec278f00 ]---
[ 1650.517577] (NULL device *): vi_capture_control_message: NULL VI channel received
[ 1650.517811] t194-nvcsi 13e10000.host1x:nvcsi@15a00000: csi5_stream_close: Error in closing stream_id=0, csi_port=0
[ 1650.518139] (NULL device *): vi_capture_control_message: NULL VI channel received
[ 1650.518381] t194-nvcsi 13e10000.host1x:nvcsi@15a00000: csi5_stream_open: VI channel not found for stream- 0 vc- 1
```

Additionally, the /dev/video0 (Depth Stream) produces the following kernel log:

```
[  479.457111] tegra-camrtc-capture-vi tegra-capture-vi: corr_err: discarding frame 0, flags: 0, err_data 131072, vc: 0
[  479.458521] (NULL device *): vi_capture_control_message: NULL VI channel received
[  479.458924] t194-nvcsi 13e10000.host1x:nvcsi@15a00000: csi5_stream_close: Error in closing stream_id=0, csi_port=0
[  479.459210] (NULL device *): vi_capture_control_message: NULL VI channel received
[  479.459395] t194-nvcsi 13e10000.host1x:nvcsi@15a00000: csi5_stream_open: VI channel not found for stream- 0 vc- 0
```
Has anyone else experienced this issue or have any insights on how to resolve it? Any help would be greatly appreciated!

-Jose

rr-jmorera, 2024-10-31:
I resolved the issue related to the backtrace by applying the following patch:

```
diff --git a/kernel/nvidia/drivers/media/platform/tegra/camera/vi/vi5_fops.c b/kernel/nvidia/drivers/media/platform/tegra/camera/vi/vi5_fops.c
index 8ba911c4d..6eb9aab2d 100644
--- a/kernel/nvidia/drivers/media/platform/tegra/camera/vi/vi5_fops.c
+++ b/kernel/nvidia/drivers/media/platform/tegra/camera/vi/vi5_fops.c
@@ -975,11 +975,12 @@ static int vi5_channel_start_streaming(struct vb2_queue *vq, u32 count)
        /* Skip in bypass mode */
        if (!chan->bypass) {
                for (vi_port = 0; vi_port < chan->valid_ports; vi_port++) {
-                       int err = vi5_channel_open(chan, vi_port);
-
-                       if (err)
+                       chan->tegra_vi_channel[vi_port] =
+                                       vi_channel_open_ex(chan->id + vi_port, false);
+                       if (IS_ERR(chan->tegra_vi_channel[vi_port])) {
+                               ret = PTR_ERR(chan);
                                goto err_open_ex;
-
+                       }
                        spin_lock_irqsave(&chan->capture_state_lock, flags);
                        chan->capture_state = CAPTURE_IDLE;
                        spin_unlock_irqrestore(&chan->capture_state_lock, flags);
@@ -1120,8 +1121,7 @@ static int vi5_channel_stop_streaming(struct vb2_queue *vq)
                                dev_err(&chan->video->dev,
                                        "vi capture release failed\n");
 
-                       vi_channel_close_ex(chan->vi_channel_id[vi_port],
-                                               chan->tegra_vi_channel[vi_port]);
+                       vi_channel_close_ex(chan->id + vi_port, chan->tegra_vi_channel[vi_port]);
                        chan->tegra_vi_channel[vi_port] = NULL;
                }
```
However, I’m still encountering this issue (in all streams). Has anyone else experienced this, or does anyone have additional suggestions for troubleshooting? Any help would be appreciated!

```
[  479.457111] tegra-camrtc-capture-vi tegra-capture-vi: corr_err: discarding frame 0, flags: 0, err_data 131072, vc: 0
[  479.458521] (NULL device *): vi_capture_control_message: NULL VI channel received
[  479.458924] t194-nvcsi 13e10000.host1x:nvcsi@15a00000: csi5_stream_close: Error in closing stream_id=0, csi_port=0
[  479.459210] (NULL device *): vi_capture_control_message: NULL VI channel received
[  479.459395] t194-nvcsi 13e10000.host1x:nvcsi@15a00000: csi5_stream_open: VI channel not found for stream- 0 vc- 0
```

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.

Thanks for the patch. If it's still needed on the current release, a PR is welcome.