realsenseai/realsense_mipi_platform_driver #246: max9295 12-0044: max9295_write_reg:i2c write failed, 0x10 = 21

chenx915, 2024-12-10:
We were trying to adapt the D457 to the Jetson ORIN NX platform, and during the process of migrating the driver, we found that the I2C address of the camera could only be queried once.

MAX9296 and INTEL realsense D457 GMSL were used
[image]
```
[    6.384383] max9296 12-004a: [MAX9296]: probing GMSL Deserializer
[    6.384431] max9296 12-004a: max9296_probe:  success
[    6.385744] max9295 12-0044: [MAX9295]: probing GMSL Serializer
[    6.385777] max9295 12-0044: max9295_probe: success
[    6.385803] max9295 12-0042: [MAX9295]: probing GMSL Serializer
[    6.385823] max9295 12-0042: max9295_probe: success
[    6.386986] d4xx 9-001a: Probing driver for D45x
[    6.387061] ----plink----ds5_board_setup----ifdef--
[    6.387065] serializer reg: 0x44
[    6.387072] state->g_ctx.csi_mode: 0x1
[    6.387074] ----plink----state->g_ctx.num_csi_lanes----: 0x2
[    6.725794] max9295 12-0044: max9295_write_reg:i2c write failed, 0x10 = 21
[    6.725911] max9295 12-0044: max9295_setup_control: ERROR: ser device not found
[    6.725913] d4xx 9-001a: gmsl serializer setup failed
[    6.728330] tegra194-pcie 141e0000.pcie: Phy link never came up
[    6.759090] d4xx 9-001a: ds5_serdes_setup gmsl serdes setup failed
[    6.759183] d4xx: probe of 9-001a failed with error -121
[    6.759222] d4xx 10-001a: Probing driver for D45x
[    6.759312] ----plink----ds5_board_setup----ifdef--
[    6.759314] serializer reg: 0x44
[    6.759326] state->g_ctx.csi_mode: 0x1
[    6.759327] ----plink----state->g_ctx.num_csi_lanes----: 0x2
[    6.759328] ----plink----cannot handle more than 24 D457 cameras----
[    6.792664] d4xx 10-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.796690] systemd-journald[250]: Received client request to flush runtime journal.
[    6.825849] d4xx 10-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.859037] d4xx 10-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.892656] d4xx 10-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.925859] d4xx 10-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.959075] d4xx 10-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.959086] d4xx 10-001a: ds5_probe(): cannot communicate with D4XX: -121 on addr: 0x1a
[    6.965197] d4xx: probe of 10-001a failed with error -121
[    6.966493] d4xx 11-001a: Probing driver for D45x
[    6.971640] ----plink----ds5_board_setup----ifdef--
[    6.971646] serializer reg: 0x44
[    6.971666] state->g_ctx.csi_mode: 0x1
[    6.971668] ----plink----state->g_ctx.num_csi_lanes----: 0x2
[    6.971669] ----plink----cannot handle more than 24 D457 cameras----
[    6.974472] d4xx 11-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.974826] d4xx 11-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.975163] d4xx 11-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.975504] d4xx 11-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.975835] d4xx 11-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.976167] d4xx 11-001a: ds5_read(): i2c read failed -121, 0x5020
[    6.976168] d4xx 11-001a: ds5_probe(): cannot communicate with D4XX: -121 on addr: 0x1a
[    6.982940] d4xx: probe of 11-001a failed with error -121
[    6.983610] d4xx 12-001a: Probing driver for D45x
[    6.985919] ----plink----ds5_board_setup----ifdef--
[    6.985925] serializer reg: 0x44
[    6.985943] state->g_ctx.csi_mode: 0x1
[    6.985945] ----plink----state->g_ctx.num_csi_lanes----: 0x2
[    6.985946] ----plink----cannot handle more than 24 D457 cameras----
[    7.020168] d4xx 12-001a: ds5_read(): i2c read failed -121, 0x5020
[    7.053388] d4xx 12-001a: ds5_read(): i2c read failed -121, 0x5020
[    7.053826] tegra-soc-hwpm f100000.hwpm: Adding to iommu group 25
[    7.078197] scare-pigeon 13e00000.host1x:vi0-thi@15f00000: Adding to iommu group 26
[    7.086559] d4xx 12-001a: ds5_read(): i2c read failed -121, 0x5020
```

chenx915, 2024-12-16:
Can someone take a look？

taeyeon-toofon, 2024-12-16:
I have a same issue and changed different D457, but it didn't work.
[image]

Arun-Prasad-V, 2024-12-18:
Hi @chenx915 . Are you using Intel's deserializer board https://store.intelrealsense.com/buy-intel-realsense-des457.html or different?
If different, I hope you would have updated the device tree accordingly.

taeyeon-toofon, 2024-12-18:
Hello @Arun-Prasad-V , Maybe my case is different from chenx915, but I am using Intel's deserializer board and I got that error. I am using Orin AGX, JetPack 6.0, and LI-JTX1-SUB-ADTP board.

Currently, I am using https://github.com/IntelRealSense/realsense_mipi_platform_driver/commit/81c2304ad01a53dcbd85a2bf08b5941f0e677306 commit. Because when I used latest dev branch code(https://github.com/IntelRealSense/realsense_mipi_platform_driver/commit/61afe61296a3d62c3eee940147e33419f7422700), I can not get any message for d4xx. Please let me know if you have any suggestions. Thank you very much.

```
nvidia@ubuntu:~$ sudo dmesg | grep tegra-capture-vi
[    9.357521] platform 13e00000.host1x:nvcsi@15a00000: Fixing up cyclic dependency with tegra-capture-vi

nvidia@ubuntu:~$ sudo dmesg | grep d4xx
[    9.443608] d4xx 9-001a: Probing driver for D45x
[    9.983168] d4xx 9-001a: ds5_chrdev_init() class_create
[    9.989521] d4xx 9-001a: D4XX Sensor: DEPTH, firmware build: 5.15.1.0
[   10.007813] d4xx 12-001a: Probing driver for D45x
[   10.013899] d4xx 12-001a: D4XX Sensor: RGB, firmware build: 5.15.1.0
[   10.025787] d4xx 13-001a: Probing driver for D45x
[   10.029095] d4xx 13-001a: D4XX Sensor: Y8, firmware build: 5.15.1.0
[   10.041282] d4xx 14-001a: Probing driver for D45x
[   10.044759] d4xx 14-001a: D4XX Sensor: IMU, firmware build: 5.15.1.0

Arun-Prasad-V, 2024-12-19:
> Maybe my case is different from chenx915, but I am using Intel's deserializer board and I got that error. I am using Orin AGX, JetPack 6.0, and LI-JTX1-SUB-ADTP board.

Hi @taeyeon-toofon, this setup is running fine from our side. If you trying it for the first time, please make sure the camera D457 is in MIPI mode (there will be a switch at the back of camera where we can switch between USB and MIPI mode).

To confirm FW is already running fine in D457 camera, you can switch to USB mode and connect it through Type-C cable. Ignore the dmesg logs for this case and open "realsense-viewer" and check if it is detected as D455 camera properly.

Also, try powering off both Jetson and deserializer and wait for a minute and power ON. Because, sometimes power cycling deserializer along with Jetson helps smooth initialization.

If you are still facing issues, please create a separate ticket (as your setup is different from this ticket) and share dmesg log.

sammychentw, 2024-12-23:
Hi, @Arun-Prasad-V :

I got same I2C write fail in Jetpack 6.0, too.
```
[Kernel dmesg log]
[   10.456682] [D457] max9296_setup_link()
[   10.456684] [D457] max9296_get_sdev_idx()
[   10.456686] [D457] max9296_write_link(link=0x00000001)
[   10.456688] [D457] max9296_write_reg(addr=0x0010,val=0x0001)
[   10.457257] max9296 2-0048: max9296_write_reg:i2c write failed, 0x10 = 1
[   10.457372] [D457] max9296_write_reg(addr=0x0010,val=0x0021)
[   10.458635] max9296 2-0048: max9296_write_reg:i2c write failed, 0x10 = 21
[   10.674991] d4xx 2-001a: gmsl serializer setup failed
```
I try to add some debug function to ds5_probe() in d4xx.c,
and I found that the max9296_write_reg() use regmap_write() to I2C write 0x48 0x10 1 and  I2C write 0x48 0x10 21,
and it will be get i2c write failed in Jetpack6.
```
#define MAX9296_CTRL0_ADDR 0x10

max9296_write_link () {
max9296_write_reg(dev, MAX9296_CTRL0_ADDR, 0x01);
max9296_write_reg(dev, MAX9296_CTRL0_ADDR, 0x21);
}

static int max9296_write_reg(struct device *dev, u16 addr, u8 val)
{
	struct max9296 *priv;
	int err;
	printk( "[D457] %s(addr=0x%04X,val=0x%04X)\n",__func__, addr, val);	// for debug by YUAN Sammy, 2024-12-20

	priv = dev_get_drvdata(dev);

	err = regmap_write(priv->regmap, addr, val);
	if (err)
		dev_err(dev, "%s:i2c write failed, 0x%x = %x\n", __func__, addr, val);

	/* delay before next i2c command as required for SERDES link */
	usleep_range(100, 110);

	return err;
}
```

If I change i2c write to d4xx.c by i2c_transfer() in Jetpack6,
and it doesn't get i2c write issue. 

Could you help to solve I2C write failed issue by regmap_write for Intel D457 in Jetpack 6?

chenx915, 2025-02-11:
> Hi [@chenx915](https://github.com/chenx915) . Are you using Intel's deserializer board https://store.intelrealsense.com/buy-intel-realsense-des457.html or different? If different, I hope you would have updated the device tree accordingly.

We are not using intel's official de-string board, but our own research, using MAX9296A.

chenx915, 2025-02-11:
In addition, as required by the company, other solutions have been replaced in the project, and D457 has not been adapted for the time being.