realsenseai/realsense_mipi_platform_driver #247: Build issues on Jetson Orin

2vin2vin, 2025-01-02:
I am using jetson orin
jetpack 6.0
after following the process mentioned on the page getting this error
mime@mime:~$./build_all_deb.sh 6.0
make[1]: Entering directory '/home/inertia_ugv/realsense_mipi_platform_driver/images/6.0'
  GEN     Makefile
***
*** Can't find default configuration "arch/arm64/configs/tegra_defconfig"!
***
make[2]: *** [/home/inertia_ugv/realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src/scripts/kconfig/Makefile:94: tegra_defconfig] Error 1
make[1]: *** [/home/inertia_ugv/realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src/Makefile:630: tegra_defconfig] Error 2
make[1]: Leaving directory '/home/inertia_ugv/realsense_mipi_platform_driver/images/6.0'
make: *** [Makefile:226: __sub-make] Error 2

I also looked into this https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/215
but I did not understand what to do

2vin2vin, 2025-01-06:
> I am using jetson orin jetpack 6.0 after following the process mentioned on the page getting this error mime@mime:~$./build_all_deb.sh 6.0 make[1]: Entering directory '/home/ugv/realsense_mipi_platform_driver/images/6.0' GEN Makefile
> 
> *** Can't find default configuration "arch/arm64/configs/tegra_defconfig"!
> 
> make[2]: *** [/home/ugv/realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src/scripts/kconfig/Makefile:94: tegra_defconfig] Error 1 make[1]: *** [/home/ugv/realsense_mipi_platform_driver/sources_6.0/kernel/kernel-jammy-src/Makefile:630: tegra_defconfig] Error 2 make[1]: Leaving directory '/home/ugv/realsense_mipi_platform_driver/images/6.0' make: *** [Makefile:226: __sub-make] Error 2
> 
> I also looked into this #215 but I did not understand what to do

2vin2vin, 2025-01-06:
> I am using jetson orin jetpack 6.0 after following the process mentioned on the page getting this error mime@mime:~$./build_all_deb.sh 6.0 make[1]: Entering directory [...]

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.

The build scripts have been reworked since. Please retry with `setup_workspace.sh` → `apply_patches.sh` → `build_all.sh` and attach the full log if it fails.