realsenseai/realsense_mipi_platform_driver #249: How to buy the Intel RealSense De-Serialize board?

sammychentw, 2025-01-20:
Hi, Intel:

We want to buy the Intel RealSense De-Serialize board, and use it to Intel RealSense D457 GMSL MIPI camera on Nvidia Jetson AGX Orin DevKit.
But the product URL can't access, how to buy this Intel RealSense De-Serialize board?
Could you provide the sales contact window to us?

RealSense™ De-Serialize board
[https://store.intelrealsense.com/buy-intel-realsense-des457.html](https://store.intelrealsense.com/buy-intel-realsense-des457.html)

[image]

sammychentw, 2025-02-04:
Hi @kevinqzhao :

Which model of FangZhu SerDes board could support Intel RealSense mipi platform driver for JP6.2 on GitHub?
https://github.com/IntelRealSense/realsense_mipi_platform_driver

kevinqzhao, 2025-02-05:
Please use https://www.fangzhutech.com/en/pd.jsp?id=12

sammychentw, 2025-02-05:
HI, @kevinqzhao :

Thanks for your information.

According to dmipx had been metion the Intel RealSense GMSL De-serializer Board, he say this board had design some schematic for Intel D457 mipi driver code on github.
[https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/166#issuecomment-1637003984](https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/166#issuecomment-1637003984)

May I to know the FG96-8CH board have some schemtic design for Intel D457 mipi driver code on github?
Fangzhu Technology Model：[FG96-8CH](https://www.fangzhutech.com/en/pd.jsp?id=12)

ymodlin, 2026-11-10:
This is a hardware purchasing question, which RealSense support handles better than this repo. Fangzhu boards are supported through the overlays listed in `README_JP6.2.md`. Closing.

closed 2026-11-10 by ymodlin