realsenseai/realsense_mipi_platform_driver #250: Realsense Mipi platform driver for jetson orin nano

Robo-Dude, 2025-01-27:
Hello Everyone,

Recently, I brought a intel realsense camera D457, I got to know that, I need to setup **realsense_mipi_platform_driver,**  but the problem is, I have **jetson Orin Nano (jetpack 6.0)**  I saw there is dev branch available while tested on jetson orin AGX, so my doubt is, **Is this driver can setup on jetson orin nano?** , if yes, can someone provide way to do that on **jetson orin nano**.

ymodlin, 2025-05-15:
We currently support the Nvidia Jetson Orin AGX. we didn't test it on the nano platform.

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.

This updates my earlier reply: Orin Nano now works on the carriers above.