realsenseai/realsense_mipi_platform_driver #255: [JetPack 6.0] RealSense viewer get xioctl(VIDIOC_G_EXT_CTRLS) failed on AGX Orin DevKit.

sammychentw, 2025-02-17:
Hi, all:

I use Intel D457 GMSL MIPI camera on Jetson AGX Orin DevKit, 
and I build the realsense_mipi_platform_driver dev branch, 
and install it to JetPack6.0.

I could get the /dev/video0~5 device on Jetson AGX Orin DevKit:
```
video-rs-depth-0 -> /dev/video0
video-rs-depth-md-0 -> /dev/video1
video-rs-color-0 -> /dev/video2
video-rs-color-md-0 -> /dev/video3
video-rs-ir-0 -> /dev/video4
video-rs-imu-0 -> /dev/video5
```

But the realsense-viewer will get the "xioctl(VIDIOC_G_EXT_CTRLS) failed" and it can't play  video.
```
[ RealSense viewer log ]
# realsense-viewer --debug
 14/02 16:41:32,846 DEBUG [281473023563840] (context.cpp:55) Librealsense VERSION: 2.55.1.0
 14/02 16:41:32,132 DEBUG [281473023563840] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 14/02 16:41:32,132 DEBUG [281473023563840] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 14/02 16:41:32,132 DEBUG [281473023563840] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 14/02 16:41:32,163 DEBUG [281473023563840] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
 14/02 16:41:32,359 INFO [281473023563840] (context.cpp:116) ... /dev/video-rs-depth-0
 14/02 16:41:32,359 INFO [281473023563840] (context.cpp:128) Found 1 RealSense devices (0xff requested & 0xff from device-mask in settings)
 14/02 16:41:32,518 INFO [281473023563840] (rs.cpp:3103) Framebuffer size changed to 2688 x 1512
 14/02 16:41:32,519 INFO [281473023563840] (rs.cpp:3103) Scale Factor is now 1
 14/02 16:41:32,570 ERROR [281472888830016] (librealsense-exception.h:52) xioctl(VIDIOC_G_EXT_CTRLS) failed Last Error: Out of streams resources
 14/02 16:41:32,571 ERROR [281472888830016] (rs.cpp:237) [rs2_create_device( info_list:0xaaaadced4500, index:0 ) Backend] xioctl(VIDIOC_G_EXT_CTRLS) failed Last Error: Out of streams resources
 14/02 16:41:32,571 ERROR [281472888830016] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
 14/02 16:41:32,571 WARNING [281472888830016] (rs.cpp:373) null pointer passed for argument "device"
 14/02 16:41:32,571 WARNING [281472888830016] (rs.cpp:3106) Couldn't refresh devices - xioctl(VIDIOC_G_EXT_CTRLS) failed Last Error: Out of streams resources
 14/02 16:41:32,817 INFO [281473023563840] (context.cpp:116) ... /dev/video-rs-depth-0
 14/02 16:41:32,817 INFO [281473023563840] (context.cpp:128) Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
```

How to know what's root cause? How to debug this VIDIOC_G_EXT_CTRLS fail?

Nir-Az, 2025-02-17:
Hi @sammychentw ,
Did you run the udev_rules script?
Sorry for the question but did you try rebooting the system inc deserializer + Jetson? :)

sammychentw, 2025-02-19:
Hi @Nir-Az ,
May I to know what is  udev_rules script?
I follow the README_JP6.md in 1.0.1.25 tag github.
[https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/r/1.0.1.25/README_JP6.md](https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/r/1.0.1.25/README_JP6.md)

I had been try rebooting the Jetson AGX Orin DevKit and deserializer, 
and it get same error in realsense-viewer.

Here my device information:
```
root@agx-orin-devit-jp60:/home/nvidia# ll /dev/d4xx-dfu-*
lrwxrwxrwx 1 root root      20 Feb 19 10:40 /dev/d4xx-dfu-0 -> /dev/d4xx-dfu-9-001a
crw-rw---- 1 root video 505, 0 Feb 19 10:39 /dev/d4xx-dfu-9-001a

root@agx-orin-devit-jp60:/home/nvidia# ls -l /dev/video*
crw-rw----+ 1 root video 81,  0 Feb 19 10:39 /dev/video0
crw-rw----+ 1 root video 81,  1 Feb 19 10:39 /dev/video1
crw-rw----+ 1 root video 81, 23 Feb 19 10:39 /dev/video2
crw-rw----+ 1 root video 81, 24 Feb 19 10:39 /dev/video3
crw-rw----+ 1 root video 81, 26 Feb 19 10:39 /dev/video4
crw-rw----+ 1 root video 81, 28 Feb 19 10:39 /dev/video5
lrwxrwxrwx  1 root root      11 Feb 19 10:40 /dev/video-rs-color-0 -> /dev/video2
lrwxrwxrwx  1 root root      11 Feb 19 10:40 /dev/video-rs-color-md-0 -> /dev/video3
lrwxrwxrwx  1 root root      11 Feb 19 10:40 /dev/video-rs-depth-0 -> /dev/video0
lrwxrwxrwx  1 root root      11 Feb 19 10:40 /dev/video-rs-depth-md-0 -> /dev/video1
lrwxrwxrwx  1 root root      11 Feb 19 10:40 /dev/video-rs-imu-0 -> /dev/video5
lrwxrwxrwx  1 root root      11 Feb 19 10:40 /dev/video-rs-ir-0 -> /dev/video4

root@agx-orin-devit-jp60:/home/nvidia# ls -l /sys/class/d4xx-class/
lrwxrwxrwx 1 root root 0 Nov 22  2023 d4xx-dfu-9-001a -> ../../devices/virtual/d4xx-class/d4xx-dfu-9-001a

root@agx-orin-devit-jp60:/home/nvidia# v4l2-ctl -d0 -C fw_version
fw_version:   84934657
```

sammychentw, 2025-02-20:
Hi, @Nir-Az :

I had setup_udev_rules.sh in AGX Orin DevKit, and see success.
```
# ./scripts/setup_udev_rules.sh 
Setting-up permissions for RealSense devices
Remove all RealSense cameras attached. Hit any key when ready
udev-rules successfully installed
```

The realsense-viewer log is change, but it still can't play video, and show belw read message: 
```
Multiple realsense udev-rules were found!:
1: /etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies
```
[realsense-viewer log ]
```
# realsense-viewer --debug
 20/02 11:48:28,712 DEBUG [281473530288192] (context.cpp:55) Librealsense VERSION: 2.55.1.0
 20/02 11:48:30,221 DEBUG [281473530288192] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 20/02 11:48:30,221 DEBUG [281473530288192] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 20/02 11:48:30,221 DEBUG [281473530288192] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 20/02 11:48:30,254 DEBUG [281473530288192] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
 20/02 11:48:30,503 INFO [281473530288192] (context.cpp:116) ... /dev/video-rs-depth-0
 20/02 11:48:30,503 INFO [281473530288192] (context.cpp:128) Found 1 RealSense devices (0xff requested & 0xff from device-mask in settings)
 20/02 11:48:30,531 INFO [281473530288192] (rs.cpp:3103) Framebuffer size changed to 2688 x 1505
 20/02 11:48:30,531 INFO [281473530288192] (rs.cpp:3103) Window size changed to 2688 x 1505
 20/02 11:48:30,531 INFO [281473530288192] (rs.cpp:3103) Scale Factor is now 1
 20/02 11:48:30,588 ERROR [281473345288256] (librealsense-exception.h:52) xioctl(VIDIOC_G_EXT_CTRLS) failed Last Error: Out of streams resources
 20/02 11:48:30,589 ERROR [281473345288256] (rs.cpp:237) [rs2_create_device( info_list:0xaaaac39e4060, index:0 ) Backend] xioctl(VIDIOC_G_EXT_CTRLS) failed Last Error: Out of streams resources
 20/02 11:48:30,589 ERROR [281473345288256] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
 20/02 11:48:30,589 WARNING [281473345288256] (rs.cpp:373) null pointer passed for argument "device"
 20/02 11:48:30,589 WARNING [281473345288256] (rs.cpp:3106) Couldn't refresh devices - xioctl(VIDIOC_G_EXT_CTRLS) failed Last Error: Out of streams resources
 20/02 11:48:30,864 INFO [281473530288192] (context.cpp:116) ... /dev/video-rs-depth-0
 20/02 11:48:30,865 INFO [281473530288192] (context.cpp:128) Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
```

I am using the D457 mipi device on AGX Orin DevKit now,
so I need to remove /etc/udev/rules.d/99-realsense-libusb.rules ? or any comment for it?

ymodlin, 2026-11-10:
Hi, we're going through old issues. The driver has changed a lot since this was reported: JetPack 6.0–6.2.1, 7.0 and 7.1 support, many stability and recovery fixes, and Orin Nano support on the Seeed GMSL and Fangzhu FG12-4CH carriers. Could you retry with the latest release from this repo? If it still happens, please attach `cat /sys/module/d4xx/version`, your JetPack version, the camera FW version, and the full `dmesg`. If we don't hear back in 30 days, we'll close this. Feel free to reopen with the data.

Your log also shows duplicate udev rules (`99-realsense-libusb.rules` + `60-librealsense2-udev-rules.rules`). Please keep only one.