realsenseai/realsense_mipi_platform_driver #284: Realsense MIPI driver for Jetpack 6.2.1

GinesGomariz, 2025-07-15:
Is this repository available with Jetson Orin Nano Developer Kit with Jetpack 6.2.1 (latest Jetpack version for Jetson Series)? 

On the other hand, I would like to ask how to execute all scripts. Every time I do it, my Jetpack Image gets corrupted and do not boot properly. 
In my case, I execute as follows (I use every time setup script mode):

**On my Computer (setup script mode):**

`- ./setup_workspace.sh 6.0
 - ./apply_patches.sh --one-cam apply 6.0
 - sudo apt install build-essential bc flex bison
 - ./build_all.sh 6.0
 - ./apply_patches.sh reset 6.0`
 
After that, I pass the generated files as guide shows:

`scp -r images/6.0/rootfs/boot/tegra234-camera-d4xx-overlay.dtbo nvidia@10.0.0.116:~/
scp -r images/6.0/rootfs/lib/modules/5.15.122-tegra/extra nvidia@10.0.0.116:~/
# RealSense metadata patched kernel modules
scp -r images/6.0/rootfs/lib/modules/5.15.122-tegra/kernel/drivers/media/v4l2-core/videodev.ko nvidia@10.0.0.116:~/
scp -r images/6.0/rootfs/lib/modules/5.15.122-tegra/kernel/drivers/media/usb/uvc/uvcvideo.ko nvidia@10.0.0.116:~/
# Kernel Image with SENSOR_HID support for RealSense USB cameras with IMU
scp -r images/6.0/rootfs/boot/Image nvidia@10.0.0.116:~/`

**On my Jetson Orin Nano DevKit:**

`# 1º Copy the passed files: 
sudo cp ~/tegra234-camera-d4xx-overlay.dtbo /boot/
# backup:
sudo tar -cjf /lib/modules/$(uname -r)/modules_$(uname -r)_extra.tar.bz2 /lib/modules/$(uname -r)/extra
sudo cp -r ~/extra /lib/modules/$(uname -r)/
# enable RealSense metadata:
sudo cp uvcvideo.ko /lib/modules/5.15.122-tegra/kernel/drivers/media/usb/uvc/uvcvideo.ko
sudo cp videodev.ko /lib/modules/5.15.122-tegra/kernel/drivers/media/v4l2-core/videodev.ko
# backup kernel (better to have additional boot entry in extlinux.conf)
sudo cp /boot/Image /boot/Image.orig
sudo cp Image /boot/Image

# 2º Enable d4xx overlay:
sudo /opt/nvidia/jetson-io/jetson-io.py # 1st error: not recognize Realsense as MIPI device, why?
# Even with this, I got an error (device not in list): 
sudo /opt/nvidia/jetson-io/config-by-hardware.py -n 2="Jetson RealSense Camera D457"

# 3º Enable d4xx autoload
echo "d4xx" | sudo tee /etc/modules-load.d/d4xx.conf
sudo depmod

# 4º Reboot:
sudo reboot
`

After all this, my Jetson does not boot and stops working (I think it is due to copy files generated with this repository into boot and drivers folders).

ymodlin, 2025-07-15:
Hi,
We currently support only the Orin AGX board.
you can try to use our code as an example but currently there is no out-of-box support for the nano environment

GinesGomariz, 2025-07-16:
And what about the Jetpack version? Is available to use Jetpack 6.2.1 or better to downgrade to 6.0?

ymodlin, 2025-07-29:
as you can see we are working on supporting the latest jetpack

closed 2026-02-26 by ymodlin